DoosanRobotics/doosan-robot #141: real模式下，客户端 T/P 身份验证失败

xwn-github, 2023-03-20:
您好，我在运行real模式时遇到问题：
我运行的指令是：roslaunch dsr_launcher single_robot_rviz.launch mode:=real host:=192.168.137.100 port:=12345 model:=a0509
DART Platform 弹出警告：客户端 T/P 身份验证失败。
[image]

xwn-github, 2023-03-20:
我使用studio控制时，是能从platform获取到控制权限的，但是运行ros时获取控制权限失败

doosan-robotics, 2023-03-20:
@xwn-github 
What environment are you using ROS in? Are you using WSL by any chance?

xwn-github, 2023-03-20:
Ubuntu20.04，ros noetic
我的电脑是Windows10，在虚拟机（VMware）z中安装的Ubuntu20.04，ros-noetic-desktop-full

doosan-robotics, 2023-04-12:
@xwn-github 
Sorry for the late reply. The virtual box port and the actual desktop port must match for normal operation. Please check the network connection status. (Try using a bridge adapter)

closed 2023-06-07 by xwn-github