DoosanRobotics/doosan-robot #142: [dsr_control] Error initializing robot

xwn-github, 2023-03-20:
运行环境：Ubuntu18.04  ros melodic
我按照melodic-devel的教程使用ros时遇到问题，我怀疑时功能包有问题，一下是报错信息：
`
xls@ubuntu:~/doosan_melodic$ roslaunch dsr_launcher single_robot_gazebo.launch mode:=virtual
... logging to /home/xls/.ros/log/02641c4e-c705-11ed-9964-000c2956cbbf/roslaunch-ubuntu-24095.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://ubuntu:36479/

SUMMARY
========

PARAMETERS
 * /dsr01/dsr_joint1_position/joint: joint1
 * /dsr01/dsr_joint1_position/pid/d: 200.0
 * /dsr01/dsr_joint1_position/pid/i: 100.0
 * /dsr01/dsr_joint1_position/pid/i_clamp: 500
 * /dsr01/dsr_joint1_position/pid/p: 500.0
 * /dsr01/dsr_joint1_position/type: position_controll...
 * /dsr01/dsr_joint2_position/joint: joint2
 * /dsr01/dsr_joint2_position/pid/d: 250.0
 * /dsr01/dsr_joint2_position/pid/i: 100.0
 * /dsr01/dsr_joint2_position/pid/i_clamp: 500
 * /dsr01/dsr_joint2_position/pid/p: 500.0
 * /dsr01/dsr_joint2_position/type: position_controll...
 * /dsr01/dsr_joint3_position/joint: joint3
 * /dsr01/dsr_joint3_position/pid/d: 40.0
 * /dsr01/dsr_joint3_position/pid/i: 20.0
 * /dsr01/dsr_joint3_position/pid/i_clamp: 500
 * /dsr01/dsr_joint3_position/pid/p: 300.0
 * /dsr01/dsr_joint3_position/type: position_controll...
 * /dsr01/dsr_joint4_position/joint: joint4
 * /dsr01/dsr_joint4_position/pid/d: 40.0
 * /dsr01/dsr_joint4_position/pid/i: 0.0
 * /dsr01/dsr_joint4_position/pid/i_clamp: 500
 * /dsr01/dsr_joint4_position/pid/p: 100.0
 * /dsr01/dsr_joint4_position/type: position_controll...
 * /dsr01/dsr_joint5_position/joint: joint5
 * /dsr01/dsr_joint5_position/pid/d: 25.0
 * /dsr01/dsr_joint5_position/pid/i: 20.0
 * /dsr01/dsr_joint5_position/pid/i_clamp: 500
 * /dsr01/dsr_joint5_position/pid/p: 200.0
 * /dsr01/dsr_joint5_position/type: position_controll...
 * /dsr01/dsr_joint6_position/joint: joint6
 * /dsr01/dsr_joint6_position/pid/d: 25.0
 * /dsr01/dsr_joint6_position/pid/i: 20.0
 * /dsr01/dsr_joint6_position/pid/i_clamp: 500
 * /dsr01/dsr_joint6_position/pid/p: 300.0
 * /dsr01/dsr_joint6_position/type: position_controll...
 * /dsr01/dsr_joint_position_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_position_controller/publish_rate: 20
 * /dsr01/dsr_joint_position_controller/type: position_controll...
 * /dsr01/dsr_joint_publisher/publish_rate: 150
 * /dsr01/dsr_joint_publisher/type: joint_state_contr...
 * /dsr01/dsr_joint_trajectory_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_trajectory_controller/publish_rate: 20
 * /dsr01/dsr_joint_trajectory_controller/type: position_controll...
 * /dsr01/dsr_velocity_controller/angular/z/has_acceleration_limits: True
 * /dsr01/dsr_velocity_controller/angular/z/has_velocity_limits: True
 * /dsr01/dsr_velocity_controller/angular/z/max_acceleration: 6.0
 * /dsr01/dsr_velocity_controller/angular/z/max_velocity: 2.0
 * /dsr01/dsr_velocity_controller/base_frame_id: base_link
 * /dsr01/dsr_velocity_controller/cmd_vel_timeout: 0.25
 * /dsr01/dsr_velocity_controller/enable_odom_tf: False
 * /dsr01/dsr_velocity_controller/estimate_velocity_from_position: False
 * /dsr01/dsr_velocity_controller/left_wheel: ['front_left_whee...
 * /dsr01/dsr_velocity_controller/linear/x/has_acceleration_limits: True
[... 155 lines cut ...]
  File "/opt/ros/melodic/lib/python2.7/dist-packages/gazebo_ros/gazebo_interface.py", line 10, in <module>
    from gazebo_msgs.srv import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/gazebo_msgs/srv/__init__.py", line 21, in <module>
    from ._SetModelState import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/gazebo_msgs/srv/_SetModelState.py", line 2, in <module>
    """autogenerated by genpy from gazebo_msgs/SetModelStateRequest.msg. Do not edit."""
KeyboardInterrupt
[INFO] [1679305986.590155, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[dsr01m1013-7] killing on exit[dsr01/drcf-5] killing on exit
[dsr01/robot_state_publisher-8] killing on exit

[dsr01/controller_spawner-4] killing on exit
[gazebo-2] killing on exit
[gazebo_gui-3] killing on exit
[dsr01m1013/controller_spawner-6] killing on exit
[WARN] [1679305986.715474, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1679305986.738577, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
xls@ubuntu:~/doosan_melodic$ 

`

ssowrira, 2023-06-06:
Hello, were you able to solve this error?
I have been having a very similar issue as well
https://github.com/doosan-robotics/doosan-robot/issues/160

Saketh-Vaishya, 2023-06-27:
Same issue.

Saketh-Vaishya, 2023-06-28:
> > 您好，您能够解决此错误吗？我也遇到了非常相似的问题#160
> > I didn't fix this either, I switched to noetic

Saketh-Vaishya, 2023-06-28:
I am using noetic , but i couldn't be able to resolve the issue