DoosanRobotics/doosan-robot #147: Error occured while installing packages

cms0373, 2023-03-24:
[image]

This error occurs when I run this command.
`sudo apt-get install ros-noetic-rqt* ros-noetic-moveit* ros-noetic-gazebo-ros-control ros-noetic-joint-state-controller ros-noetic-effort-controllers ros-noetic-position-controllers ros-noetic-ros-controllers ros-noetic-ros-control ros-noetic-joint-state-publisher-gui ros-noetic-joint-state-publisher`

When I move on to the next stage which installs moveit commander with below command,
`sudo apt-get install ros-noetic-moveit-commander`
this error occurs. It says there's unmet dependencies.
[image]

How could I fix this problem?

pucciland95, 2023-04-04:
Hi, 

A master student of mine had the same problem. 

The main problem relies in MoveIt package. Instead of typing:

Sudo apt install ros-noetic-moveit*

Just write: 

Sudo apt install ros-noetic-moveit

You just need to remove the *. I hope it helps. 

Niccolo

doosan-robotics, 2023-04-10:
We have found some issues and are currently fixing them. We will quickly take action and announce it again.