DoosanRobotics/doosan-robot #152: Publish /joint_states without taking control of the robot

dave992, 2023-04-06:
I would like to monitor the `/joint_states` of a Doosan M1013 regardless of the mode or status of the robot. 

The examples provided by this repository all seem to take motion control of the cobot as soon as the connection is established. This is preventing me from using the cobot in compliance mode for example or any other manual mode while monitoring/visualizing the cobot. 

It is possible to "take back control" via the teach pendant after connecting with the driver, but as soon as I do the `/joint_states` topic does not update correctly anymore.  I keep receiving new `joint_state` messages, but the values remain static and do not follow the current joint state values until I relaunch the node. 

Is there a simple way to publish `joint_states` regardless of the mode or status of the robot?

pucciland95, 2023-04-07:
Hi, 

I don't think I get your question. 

For sure it is not possible to "unlock" the TP when you are launching the ROS nodes (probably for safety reasons) 

However, I am 100% sure you can monitor the /joint_state topic while in manual mode. For example in my case I do "Teaching By Demonstrations" so I need to monitor the robot joints while I manually move it pressing the free drive button. 

Maybe you are confused since when you launch the robot with ROS it automatically goes in Automatic mode and the free drive motion is disabled. You can go back to manual mode with a particular service. 

Let me know if this fixes your problem or I misunderstood you question.

dave992, 2023-04-08:
> For sure it is not possible to "unlock" the TP when you are launching the ROS nodes (probably for safety reasons)

I understand, multiple points of control are not a good thing if it comes to motion 😄. I just do not want the ROS driver to be in control of any motion, yet it automatically takes control as soon as the driver is launched and I could not find any node parameters/settings that disable or change this behavior.

> Maybe you are confused since when you launch the robot with ROS it automatically goes in Automatic mode and the free drive motion is disabled. You can go back to manual mode with a particular service.

That is indeed the behavior I would like to disable. If there is a service that lets me switch back to manual mode and use the free drive, that would be a suitable workaround I would say. I'll have a look at the services then and test this early next week.

pucciland95, 2023-04-08:
The service should be called `set_robot_mode`. Passing `0` as argument should set the robot to manual mode and you can jog it through free drive.

Let me know if it works 😀

dave992, 2023-04-11:
Just tested it and it works. Thank you for the help!

It would've been nice if changing the state of the robot would have been a more active decision than just connecting with the hardware, but that is probably off-topic for this issue. 

Closing this issue now :)

closed 2023-04-11 by dave992