DoosanRobotics/doosan-robot #153: how to compile doosan driver in Nvidia-jetson-orin? Does support ARM architecture

rajDFL, 2023-04-18:
How to compile the doosan-robot package in nividia-jetson-orin that has ARM architecture
Below Error occurs 

[100%] Linking CXX executable catkin_ws/devel/lib/dsr_control/dsr_control_node
/usr/bin/ld: skipping incompatible ~/dsr_control/../common/lib/noetic/x86_64/libDRFL.a when searching for -lDRFL
/usr/bin/ld: cannot find -lDRFL
/usr/bin/ld: cannot find -lDRFL
collect2: error: ld returned 1 exit status

skyrimax, 2023-04-19:
I ran into the same issue a couple months ago. From my understanding, libDRFL.a is a library precompiled in x86_64 and therefore cannot be linked against an arm64 program. According to Doosan in the issue #58, there are no plans to support arm64, but I hope that has changed in the last 2 years.

doosan-robotics, 2023-04-28:
We do not support the ARM architecture because API-DRFL only works on x86_64. There are currently no plans to support the ARM architecture.

closed 2023-05-11 by doosan-robotics