DoosanRobotics/doosan-robot #157: Unable to shift movel()/posx() points from end effector frame to tool tip frame

DamAnirban, 2023-05-18:
Hi All,

I have a Doosan A0509s arm, and the entire setup is working fine with ROS.

However, IK for a target pose of the end-effector link is the end point of the robot (link 6 - point of attachment of the tool to the arm) when using movel() or posx(). What I need is to plan the goal in the "tool frame", for example, a frame located at the tool tip (or gripper tip, doesnt matter). 

I can define this tool frame in the URDF and with a desired pose for the tool frame (say relative to the link 6 frame), but that is clearly changing only the outer visual and not the actual kinematics. How do I "back transform" this into a target pose for the end-effector link? Or is there any other method to do this?

What I tried - I tried changing the dimensions of the link 6 itself (replaced it with a simple cylinder with length = link 6 length + tool length), to see if the end point will move outward automatically, but it did not.

Any help will be very helpful. Thanks.

AnupamKumarPranav, 2023-05-19:
Facing similar issue.. Please suggest alternative solutions.

foreverbala, 2023-07-07:
I think you have to add your tool in the teach pendant end effector tab. Then the TCP of the robot will changed to tool tip. Then when you use get_current_posx() command you will get the values along with the tool tip. The robot will move according to the tcp.