DoosanRobotics/doosan-robot #159: Error: [dsr01/spawn_create_model-10] process has finished cleanly

mariadelmar2497, 2023-06-02:
I followed all the steps for the installation and start-up. But when I try to launch the simulation on Gazebo with the following command:
$ roslaunch dsr_launcher single_robot_gazebo.launch mode:=virtual

This error occurs:

Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://158.42.59.180:45677/

SUMMARY
========

PARAMETERS
 * /dsr01/dsr_joint1_position/joint: joint1
 * /dsr01/dsr_joint1_position/pid/d: 200.0
 * /dsr01/dsr_joint1_position/pid/i: 100.0
 * /dsr01/dsr_joint1_position/pid/i_clamp: 500
 * /dsr01/dsr_joint1_position/pid/p: 500.0
 * /dsr01/dsr_joint1_position/type: position_controll...
 * /dsr01/dsr_joint2_position/joint: joint2
 * /dsr01/dsr_joint2_position/pid/d: 250.0
 * /dsr01/dsr_joint2_position/pid/i: 100.0
 * /dsr01/dsr_joint2_position/pid/i_clamp: 500
 * /dsr01/dsr_joint2_position/pid/p: 500.0
 * /dsr01/dsr_joint2_position/type: position_controll...
 * /dsr01/dsr_joint3_position/joint: joint3
 * /dsr01/dsr_joint3_position/pid/d: 40.0
 * /dsr01/dsr_joint3_position/pid/i: 20.0
 * /dsr01/dsr_joint3_position/pid/i_clamp: 500
 * /dsr01/dsr_joint3_position/pid/p: 300.0
 * /dsr01/dsr_joint3_position/type: position_controll...
 * /dsr01/dsr_joint4_position/joint: joint4
 * /dsr01/dsr_joint4_position/pid/d: 40.0
 * /dsr01/dsr_joint4_position/pid/i: 0.0
 * /dsr01/dsr_joint4_position/pid/i_clamp: 500
 * /dsr01/dsr_joint4_position/pid/p: 100.0
 * /dsr01/dsr_joint4_position/type: position_controll...
 * /dsr01/dsr_joint5_position/joint: joint5
 * /dsr01/dsr_joint5_position/pid/d: 25.0
 * /dsr01/dsr_joint5_position/pid/i: 20.0
 * /dsr01/dsr_joint5_position/pid/i_clamp: 500
 * /dsr01/dsr_joint5_position/pid/p: 200.0
 * /dsr01/dsr_joint5_position/type: position_controll...
 * /dsr01/dsr_joint6_position/joint: joint6
 * /dsr01/dsr_joint6_position/pid/d: 25.0
 * /dsr01/dsr_joint6_position/pid/i: 20.0
 * /dsr01/dsr_joint6_position/pid/i_clamp: 500
 * /dsr01/dsr_joint6_position/pid/p: 300.0
 * /dsr01/dsr_joint6_position/type: position_controll...
 * /dsr01/dsr_joint_position_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_position_controller/publish_rate: 20
 * /dsr01/dsr_joint_position_controller/type: position_controll...
 * /dsr01/dsr_joint_publisher/publish_rate: 150
 * /dsr01/dsr_joint_publisher/type: joint_state_contr...
 * /dsr01/dsr_joint_trajectory_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_trajectory_controller/publish_rate: 20
 * /dsr01/dsr_joint_trajectory_controller/type: position_controll...
 * /dsr01/dsr_velocity_controller/angular/z/has_acceleration_limits: True
 * /dsr01/dsr_velocity_controller/angular/z/has_velocity_limits: True
 * /dsr01/dsr_velocity_controller/angular/z/max_acceleration: 6.0
 * /dsr01/dsr_velocity_controller/angular/z/max_velocity: 2.0
 * /dsr01/dsr_velocity_controller/base_frame_id: base_link
 * /dsr01/dsr_velocity_controller/cmd_vel_timeout: 0.25
 * /dsr01/dsr_velocity_controller/enable_odom_tf: False
 * /dsr01/dsr_velocity_controller/estimate_velocity_from_position: False
 * /dsr01/dsr_velocity_controller/left_wheel: ['front_left_whee...
 * /dsr01/dsr_velocity_controller/linear/x/has_acceleration_limits: True
 * /dsr01/dsr_velocity_controller/linear/x/has_velocity_limits: True
[... 254 lines cut ...]
[rospy.init][INFO] 2023-06-02 10:29:53,458: registered with master
[rospy.rosout][INFO] 2023-06-02 10:29:53,458: initializing /rosout core topic
[rospy.rosout][INFO] 2023-06-02 10:29:53,462: connected to core topic /rosout
[rospy.simtime][INFO] 2023-06-02 10:29:53,463: initializing /clock core topic
[rospy.simtime][INFO] 2023-06-02 10:29:53,464: connected to core topic /clock
[rosout][INFO] 2023-06-02 10:29:53,466: Loading model XML from ros parameter /dsr01/robot_description
[rosout][INFO] 2023-06-02 10:29:53,471: Waiting for service /gazebo/spawn_urdf_model
[rospy.internal][INFO] 2023-06-02 10:29:53,737: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2023-06-02 10:29:54,541: topic[/clock] adding connection to [http://158.42.59.180:34517/], count 0
[rosout][INFO] 2023-06-02 10:29:54,676: Calling service /gazebo/spawn_urdf_model
[rosout][INFO] 2023-06-02 10:29:54,931: Spawn status: SpawnModel: Successfully spawned entity
[rospy.core][INFO] 2023-06-02 10:29:54,933: signal_shutdown [atexit]
[rospy.internal][INFO] 2023-06-02 10:29:54,934: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2023-06-02 10:29:54,934: topic[/clock] removing connection to http://158.42.59.180:34517/
[rospy.impl.masterslave][INFO] 2023-06-02 10:29:54,934: atexit
[INFO] [1685694593.466340, 0.000000]: Loading model XML from ros parameter /dsr01/robot_description
[INFO] [1685694593.471742, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1685694594.677005, 0.185000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1685694594.931727, 0.371000]: Spawn status: SpawnModel: Successfully spawned entity

So I don't understand why it is happening...

I would really appreciate any help you can provide.

Thank you in advance!

ssowrira, 2023-06-27:
Hi, were you able to rectify this problem in ROS Melodic?

duygiang1162, 2024-03-27:
How to fix this problem?