DoosanRobotics/doosan-robot #160: [ERROR] - [dsr_control] Error initializing robot (ROS Melodic)

ssowrira, 2023-06-06:
I followed the instructions under the **'melodic-devel'** branch for my **ROS Melodic** on Ubuntu 18.04 system.

when running:
`roslaunch dsr_launcher single_robot_gazebo.launch mode:=virtual`

Throws following errors:

[image]

[image]

[image]

**'noetic-devel' branch was successful in ROS Noetic on Ubuntu 20.04, however I need it to work in ROS Melodic on Ubuntu 18.04**
Any help on rectifying this issue will be appreciated.

Saketh-Vaishya, 2023-06-27:
You need to source the bash file before executing the Ros launcher

Saketh-Vaishya, 2023-06-28:
Yes might be , I am using noetic and I am getting it properly

Saketh-Vaishya, 2023-06-28:
@ssowrira  Could you able to execute real mode, because i could be able to execute virtual but i am getting same error with real mode. After reading some issues , they said that they are using network switch

DYK0905, 2023-06-29:
I have the same issue. Please correct this issue ASAP Doosan.

doosan-robotics, 2024-02-02:
Hello @Saketh-Vaishya @DYK0905 

Sorry for the late reply. We no longer support the melodic version. To control the Doosan robot with ROS in 18.04, please use Docker, or use the noetic version in the 20.04 environment. Thank you for your understanding.