DoosanRobotics/doosan-robot #164: Connecting through virtual machine

jwwood17, 2023-07-28:
Hello,
I'm trying to connect to an a0509s robot arm using ros noetic on a virtual machine running Ubuntu 20.04, but when I launch any of the scripts like

`roslaunch dsr_launcher single_robot_rviz.launch mode:=real host:=206.207.51.90 model:=a0509`
`roslaunch dsr_control dsr_control.launch mode:=real host:=206.207.51.90 model:=a0509`

I just get an authentication error on the teach pendant instead of the control passing dialog.
'Error 1.3004 Client T/P Authentication Failed'

I found these two issues which seem to indicate that it might be because of the vm.
[https://github.com/doosan-robotics/doosan-robot/issues/68](url)
[https://github.com/doosan-robotics/doosan-robot2/issues/14](url)
Are there any solutions to this issue, or does it just not work on virtual machines? Can anyone give any advice on the port manipulation that is suggested in the second linked issue?
Thank you for you time.

doosan-robotics, 2024-02-02:
Hello @jwwood17 , 

Sorry for the late reply. IP Address of doosan robotics is set on 192.168.127.100 and port number:12345. Please try this address

jwwood17, 2024-02-02:
I have set the ip address correctly through the teach pendant, and I can connect to the robot using ros on a linux computer. I'm just having an issue when trying to connect through the virtual machine on a windows computer.

xwn-github, 2024-03-20:
When running in a virtual machine, you need to set the network configuration to bridge mode