DoosanRobotics/doosan-robot #165: rviz상에서 robot 정보 오류

Jeong-bhin, 2023-08-07:
hello.

I would like to seek help with an error that the robot form cannot be recognized on rviz.

DRCF was executed with root authority and the following problem occurred.

[image]

**The environment I'm currently using is:**
- ubuntu version : 20.04
- ros version : noetic
- robot model : m1509
- python : Python 3.8.10

**The error code is:**

mbr@mbr-System-Product-Name:~/catkin_ws$ roslaunch dsr_launcher single_robot_rviz.launch model:=m1509 mode:=real host:=192.168.137.50 port:=12345
... logging to /home/mbr/.ros/log/d11519da-3505-11ee-a8ed-95cd31c59434/roslaunch-mbr-System-Product-Name-21091.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://localhost:42967/

SUMMARY
========

PARAMETERS
 * /dsr01/dsr_joint1_position/joint: joint1
 * /dsr01/dsr_joint1_position/pid/d: 200.0
 * /dsr01/dsr_joint1_position/pid/i: 100.0
 * /dsr01/dsr_joint1_position/pid/i_clamp: 500
 * /dsr01/dsr_joint1_position/pid/p: 500.0
 * /dsr01/dsr_joint1_position/type: position_controll...
 * /dsr01/dsr_joint2_position/joint: joint2
 * /dsr01/dsr_joint2_position/pid/d: 250.0
 * /dsr01/dsr_joint2_position/pid/i: 100.0
 * /dsr01/dsr_joint2_position/pid/i_clamp: 500
 * /dsr01/dsr_joint2_position/pid/p: 500.0
 * /dsr01/dsr_joint2_position/type: position_controll...
 * /dsr01/dsr_joint3_position/joint: joint3
 * /dsr01/dsr_joint3_position/pid/d: 40.0
 * /dsr01/dsr_joint3_position/pid/i: 20.0
 * /dsr01/dsr_joint3_position/pid/i_clamp: 500
 * /dsr01/dsr_joint3_position/pid/p: 300.0
 * /dsr01/dsr_joint3_position/type: position_controll...
 * /dsr01/dsr_joint4_position/joint: joint4
 * /dsr01/dsr_joint4_position/pid/d: 40.0
 * /dsr01/dsr_joint4_position/pid/i: 0.0
 * /dsr01/dsr_joint4_position/pid/i_clamp: 500
 * /dsr01/dsr_joint4_position/pid/p: 100.0
 * /dsr01/dsr_joint4_position/type: position_controll...
 * /dsr01/dsr_joint5_position/joint: joint5
 * /dsr01/dsr_joint5_position/pid/d: 25.0
 * /dsr01/dsr_joint5_position/pid/i: 20.0
 * /dsr01/dsr_joint5_position/pid/i_clamp: 500
 * /dsr01/dsr_joint5_position/pid/p: 200.0
 * /dsr01/dsr_joint5_position/type: position_controll...
 * /dsr01/dsr_joint6_position/joint: joint6
 * /dsr01/dsr_joint6_position/pid/d: 25.0
 * /dsr01/dsr_joint6_position/pid/i: 20.0
 * /dsr01/dsr_joint6_position/pid/i_clamp: 500
 * /dsr01/dsr_joint6_position/pid/p: 300.0
 * /dsr01/dsr_joint6_position/type: position_controll...
 * /dsr01/dsr_joint_position_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_position_controller/publish_rate: 20
 * /dsr01/dsr_joint_position_controller/type: position_controll...
 * /dsr01/dsr_joint_publisher/publish_rate: 150
 * /dsr01/dsr_joint_publisher/type: joint_state_contr...
[... 148 lines cut ...]
[ INFO] [1691401692.651198268]:     DRCF version = GF02100001
[ INFO] [1691401692.651211324]:     DRFL version = GL010118
[ INFO] [1691401692.651223551]:     m_nVersionDRCF = 2100001
[ INFO] [1691401692.651235275]: _______________________________________________
[ INFO] [1691401693.872500001]: On Monitor State
[callback OnMonitoringStateCB] current state: (1) STANDBY
[ INFO] [1691401693.894411744]: [init]::read 0-pos:   0.000
[ INFO] [1691401693.894493313]: [init]::read 1-pos:   0.000
[ INFO] [1691401693.894523230]: [init]::read 2-pos:   0.000
[ INFO] [1691401693.894547102]: [init]::read 3-pos:   0.000
[ INFO] [1691401693.894577271]: [init]::read 4-pos:   0.000
[ INFO] [1691401693.894600731]: [init]::read 5-pos:   0.000
[ INFO] [1691401693.923708850]: [dsr_control] controller_manager is updating!
[INFO] [1691401693.976694]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1691401693.983685]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1691401693.989229]: Loading controller: dsr_joint_position_controller
[INFO] [1691401694.004514]: Loading controller: dsr_joint_publisher
[INFO] [1691401694.024532]: Controller Spawner: Loaded controllers: dsr_joint_position_controller, dsr_joint_publisher
[INFO] [1691401694.034509]: Started controllers: dsr_joint_position_controller, dsr_joint_publisher
[WARN] [1691401722.368508]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[dsr01/controller_spawner-1] process has finished cleanly
log file: /home/mbr/.ros/log/d11519da-3505-11ee-a8ed-95cd31c59434/dsr01-controller_spawner-1*.log

**MoveitCommander works fine, but when I launch it in rviz, I keep getting errors.
Any help would be appreciated!**

Jeong-bhin, 2023-08-24:
> 혹시 해결하셨나요??

아니요 해결하지 못했습니다

griffinkang, 2023-11-14:
혹시 해결하셨나요?

doosan-robotics, 2024-02-02:
안녕하세요 @Jeong-bhin 님

답변이 늦어져서 죄송합니다.
현재 해당 문제 대한 이슈를 해결한 버전으로 2월 말에 릴리즈 계획 중에 있습니다.
조금만 양해 부탁드립니다.

song-ms, 2024-03-13:
@Jeong-bhin @fpdlwhfwm 님

해당 이슈를 해결하여 Main 브랜치에 업데이트 되었음을 알려드립니다. 
이에 따른 설치 가이드가 다소 변경되었으니 유의하여 확인 부탁드립니다.

vyasakash231, 2024-04-03:
still getting the same error.