DoosanRobotics/doosan-robot #169: E-series model

buswayne, 2023-09-16:
Hello, it's more of a question than an issue.
Are you planning to introduce the description for the new E-series model? If so, do you have an estimated release date?
Thanks

axonv-rok, 2023-09-27:
Same questions here. 

Can A0509 configuration be used for E0509?

doosan-robotics, 2024-02-02:
Hello @buswayne @axonv-rok 

Currently, development related to the e-series is in progress and is scheduled to be released soon. Until release, we ask that you use a0509 instead. Thank you for your understanding

pucciland95, 2024-03-04:
I think they just integrated it: #185

song-ms, 2024-03-13:
@pucciland95 
Thank you for answering

@buswayne @axonv-rok 
Currently, E-series model has been added to the main branch. Please confirm the main branch. 
I will proceed to close this issue. 

Thank you

closed 2024-03-13 by doosan-robotics