DoosanRobotics/doosan-robot #171: Joint limit exceeds the maximum limit while using movel, amovel

foreverbala, 2023-10-16:
@doosan-robotics 
A0509 cobot.
I am using movel and amovel for motion.In this linear motion the 6th joint exceeds the maximum limit and the robot servo turned off.In teach pendant the maximum is set for -360 to 360.
The 6th joint moving beyond the maximum limits.It moved upto 371 degrees and stops.
Can you explain the reason of the issue and how to solve this issue?
[image]

bishwajit-kr, 2023-10-16:
you can turn on servo by using ROS service #139  . and then c
heck current `posj` . and see if it still exceeds the joint limit

foreverbala, 2023-10-17:
@bishwajit-kr 
I can't turn on the servo because the servo exceeds the joint limit. I want to go to recovery or backdrive mode to turn on the servo.My question is why the servo moves beyond the max limit in linear motion.