DoosanRobotics/doosan-robot #175: Speed scalings in moveit config packages are not working

bxtbold, 2023-11-02:
Velocity and acceleration scalings from `joint_limits.yaml` are not working.

I have inspected `dsr_control` and found out that the driver uses a trajectory callback function to send position commands instead of using PositionJointInterface, which seems to cause this issue.

Is it possible to fix the driver to fix this issue?

doosan-robotics, 2024-02-02:
Hello @bxtbold ,

We are aware of the problem. Therefore, we plan to release an improved version of this part in the soon-to-be-updated noetic package. Thank you for your understanding.

bxtbold, 2024-02-02:
Thank you for your response. Looking forward to using the updated package.
Please mention this issue to a PR or commit when it is resolved, so others who are struggling with the same issue can look up :)

doosan-robotics, 2024-03-11:
Hello @bxtbold ,

Thank you for your patience. I've updated the joint limit feature and added it. Please review it on the main branch.

closed 2025-01-16 by bxtbold