DoosanRobotics/doosan-robot #176: Robot collision does not cover the whole body

bxtbold, 2023-11-03:
As shown in the picture below, the robot collision does not cover the robot visual. Generally, the collision body should cover the whole body, so planners can generate a trajectory that avoids collision from scene objects.

I kindly request make a little effort to add few more cylinders or modify existing ones those covers the whole body.

[image]

doosan-robotics, 2024-02-28:
Dear @bxtbold,

We have updated collision body now. please check "main" branch

bxtbold, 2024-03-08:
Thank you very much. I was able to check that the issue was resolved in #185

closed 2024-03-08 by bxtbold