DoosanRobotics/doosan-robot #179: Support for Ros One, Ubuntu Jammy 22.04

simonschmeisser, 2023-11-23:
We have migrated our software from ROS Melodic to ROS One and would love to use Doosan robots here as well. Unfortunately `dsr_control_node` crashes right after the start:

```
#0  0x00007f8d9a9a74ed in Poco::Net::SocketImpl::getOption(int, int, int&) () from /lib/x86_64-linux-gnu/libPocoNet.so.80
#1  0x00007f8d9a9a75cc in Poco::Net::SocketImpl::getReceiveBufferSize() () from /lib/x86_64-linux-gnu/libPocoNet.so.80
#2  0x000055795cb4fa4b in DRAFramework::CNDKSocket::GetAddress(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, unsigned int&) ()
#3  0x000055795cb455a5 in DRAFramework::CNDKHandler::SendAuthData() ()
#4  0x000055795cb3d960 in _OpenConnection ()
#5  0x000055795ca65680 in DRAFramework::CDRFLEx::open_connection (usPort=12345, strIpAddr="192.168.10.127", this=<optimized out>)
    at /usr/include/c++/11/bits/basic_string.h:194
#6  dsr_control::DRHWInterface::init (this=0x55795e6682f0) at ../../../Entwicklung/extern/doosan-robot/dsr_control/src/dsr_hw_interface.cpp:976
#7  main (argc=<optimized out>, argv=<optimized out>) at ../../../Entwicklung/extern/doosan-robot/dsr_control/src/dsr_control_node.cpp:52
```

I assume this is due to binary incompatibility between Poco 1.11.0 (libpoconet80) on Jammy and 1.9.2 (libpoconet62) on Focal/Noetic but I might be mistaken.

Could you please provide a version of `libDRFL.a` compiled on Ubuntu 22.04? Or open source it? Or give us access to the code under a NDA?

Thanks a lot
Simon

gavanderhoorn, 2024-01-09:
https://github.com/doosan-robotics/API-DRFL/issues/31, https://github.com/doosan-robotics/API-DRFL/issues/11

simonschmeisser, 2024-01-09:
This can actually be worked around by installing `libpoco-dev` and the poco62 libraries from Ubuntu Focal in the container where you build this package. Or maybe by having a local rosdep file (?). At runtime `libpoconet62` and `libpoconet80` can be installed side by side.

gavanderhoorn, 2024-01-09:
Yes, until you want/need to link the lib against a newer version of `libpoco` for whatever reason :)

It's really rather a work-around.