DoosanRobotics/doosan-robot #181: Error with Compliance Control

jwwood17, 2023-11-27:
I am trying to use the compliance control mode to press a tool into a screw. As I understand it, this is accomplished with the following lines of code.
```
stiffness = [3000.0, 3000.0, 3000.0, 200.0, 200.0, 200.0]
force_values = [0, 0, -0.1, 0, 0, 0]
direction = [0, 0, 1, 0, 0, 0]
task_compliance_ctrl(stiffness)
set_desired_force(force_values,direction)
```

followed by
```
release_force()
release_compliance_ctrl()
```
However, sometimes when moving out of the screw afterwards, the robot jerks suddenly and throws this error
`cannot signal semaphore count would exceed maximum`
The robot doesn't respond until a full reset is performed.

doosan-robotics, 2024-02-16:
Hello @jwwood17 

This error occurs due to collision detection. Try setting the collision sensitivity to 1 in TP when performing that motion again; it should be possible. Alternatively, you can prevent collision detection by setting up a safety zone in the workspace before using that function.

Thank you

jwwood17, 2024-05-01:
> Hello @jwwood17
>
> This error occurs due to collision detection. Try setting the collision sensitivity to 1 in TP when performing that motion again; it should be possible. Alternatively, you can [...]

@doosan-robotics 
Is it possible to activate these settings via a script so I can reduce the collision sensitivity during the specific motion and then increase it again afterward for safety purposes?
Are there commands for activating and deactivating predefined safety zones?

Thank you