DoosanRobotics/doosan-robot #183: DRCF ERROR

summerery7, 2024-02-15:
.. logging to /home/summerer/.ros/log/b9443d70-cc3d-11ee-b7e6-8f7a123c72a7/roslaunch-summerer-932030.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://summerer:39131/

SUMMARY
========

PARAMETERS
 * /dsr01/dsr_joint1_position/joint: joint1
 * /dsr01/dsr_joint1_position/pid/d: 200.0
 * /dsr01/dsr_joint1_position/pid/i: 100.0
 * /dsr01/dsr_joint1_position/pid/i_clamp: 500
 * /dsr01/dsr_joint1_position/pid/p: 500.0
 * /dsr01/dsr_joint1_position/type: position_controll...
 * /dsr01/dsr_joint2_position/joint: joint2
 * /dsr01/dsr_joint2_position/pid/d: 250.0
 * /dsr01/dsr_joint2_position/pid/i: 100.0
 * /dsr01/dsr_joint2_position/pid/i_clamp: 500
 * /dsr01/dsr_joint2_position/pid/p: 500.0
 * /dsr01/dsr_joint2_position/type: position_controll...
 * /dsr01/dsr_joint3_position/joint: joint3
 * /dsr01/dsr_joint3_position/pid/d: 40.0
 * /dsr01/dsr_joint3_position/pid/i: 20.0
 * /dsr01/dsr_joint3_position/pid/i_clamp: 500
 * /dsr01/dsr_joint3_position/pid/p: 300.0
 * /dsr01/dsr_joint3_position/type: position_controll...
 * /dsr01/dsr_joint4_position/joint: joint4
 * /dsr01/dsr_joint4_position/pid/d: 40.0
 * /dsr01/dsr_joint4_position/pid/i: 0.0
 * /dsr01/dsr_joint4_position/pid/i_clamp: 500
 * /dsr01/dsr_joint4_position/pid/p: 100.0
 * /dsr01/dsr_joint4_position/type: position_controll...
 * /dsr01/dsr_joint5_position/joint: joint5
 * /dsr01/dsr_joint5_position/pid/d: 25.0
 * /dsr01/dsr_joint5_position/pid/i: 20.0
 * /dsr01/dsr_joint5_position/pid/i_clamp: 500
 * /dsr01/dsr_joint5_position/pid/p: 200.0
 * /dsr01/dsr_joint5_position/type: position_controll...
 * /dsr01/dsr_joint6_position/joint: joint6
 * /dsr01/dsr_joint6_position/pid/d: 25.0
 * /dsr01/dsr_joint6_position/pid/i: 20.0
 * /dsr01/dsr_joint6_position/pid/i_clamp: 500
 * /dsr01/dsr_joint6_position/pid/p: 300.0
 * /dsr01/dsr_joint6_position/type: position_controll...
 * /dsr01/dsr_joint_position_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_position_controller/publish_rate: 20
 * /dsr01/dsr_joint_position_controller/type: position_controll...
 * /dsr01/dsr_joint_publisher/publish_rate: 150
 * /dsr01/dsr_joint_publisher/type: joint_state_contr...
 * /dsr01/dsr_joint_trajectory_controller/joints: ['joint1', 'joint...
 * /dsr01/dsr_joint_trajectory_controller/publish_rate: 20
 * /dsr01/dsr_joint_trajectory_controller/type: position_controll...
 * /dsr01/dsr_velocity_controller/angular/z/has_acceleration_limits: True
 * /dsr01/dsr_velocity_controller/angular/z/has_velocity_limits: True
 * /dsr01/dsr_velocity_controller/angular/z/max_acceleration: 6.0
 * /dsr01/dsr_velocity_controller/angular/z/max_velocity: 2.0
 * /dsr01/dsr_velocity_controller/base_frame_id: base_link
 * /dsr01/dsr_velocity_controller/cmd_vel_timeout: 0.25
 * /dsr01/dsr_velocity_controller/enable_odom_tf: False
 * /dsr01/dsr_velocity_controller/estimate_velocity_from_position: False
 * /dsr01/dsr_velocity_controller/left_wheel: ['front_left_whee...
 * /dsr01/dsr_velocity_controller/linear/x/has_acceleration_limits: True
 * /dsr01/dsr_velocity_controller/linear/x/has_velocity_limits: True
 * /dsr01/dsr_velocity_controller/linear/x/max_acceleration: 3.0
 * /dsr01/dsr_velocity_controller/linear/x/max_velocity: 1.0
 * /dsr01/dsr_velocity_controller/pose_covariance_diagonal: [0.001, 0.001, 0....
[... 126 lines cut ...]
log file: /home/summerer/.ros/log/b9443d70-cc3d-11ee-b7e6-8f7a123c72a7/dsr01m0609-6*.log
[INFO] [1708029214.910918, 1614.478000]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1708029214.912214, 1614.479000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1708029214.913571, 1614.480000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1708029214.915216, 1614.482000]: Loading controller: dsr_joint_position_controller
[INFO] [1708029214.918880, 1614.486000]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1708029214.919674, 1614.487000]: Controller Spawner: Loaded controllers: dsr_joint_position_controller
[INFO] [1708029214.921731, 1614.489000]: Started controllers: dsr_joint_position_controller
[ERROR] [1708029214.948697009, 1614.516000000]: SpawnModel: Failure - model name dsr01 already exist.
[ERROR] [1708029214.949568, 1614.516000]: Spawn service failed. Exiting.
[gazebo-1] process has died [pid 932046, exit code 255, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode /home/summerer/catkin_ws/src/doosan-robot/dsr_launcher/worlds/empty.world __name:=gazebo __log:=/home/summerer/.ros/log/b9443d70-cc3d-11ee-b7e6-8f7a123c72a7/gazebo-1.log].
log file: /home/summerer/.ros/log/b9443d70-cc3d-11ee-b7e6-8f7a123c72a7/gazebo-1*.log
[ INFO] [1708029215.023663614]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1708029215.024253672]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[dsr01/spawn_create_model-9] process has died [pid 932061, exit code 1, cmd /opt/ros/noetic/lib/gazebo_ros/spawn_model -x 0 -y 0 -z 0 -R 0 -P 0 -Y 0 -urdf -param /dsr01/robot_description -model dsr01 __name:=spawn_create_model __log:=/home/summerer/.ros/log/b9443d70-cc3d-11ee-b7e6-8f7a123c72a7/dsr01-spawn_create_model-9.log].
log file: /home/summerer/.ros/log/b9443d70-cc3d-11ee-b7e6-8f7a123c72a7/dsr01-spawn_create_model-9*.log
[WARN] [1708029244.999746, 1644.557000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[dsr01m0609/controller_spawner-5] process has finished cleanly
log file: /home/summerer/.ros/log/b9443d70-cc3d-11ee-b7e6-8f7a123c72a7/dsr01m0609-controller_spawner-5*.log

common/bin/DRCF에 들어가서 sudo ./DRCF64 12345를 입력하면
summerer@summerer:~/catkin_ws/src/doosan-robot/common/bin/DRCF$ sudo ./DRCF64
sudo: ./DRCF64: 명령이 없습니다
이렇게 뜨고
chmod +x로 권한을 추가해도 오류가 해결되지 않습니다.

doosan-robotics, 2024-02-16:
안녕하세요 @summerery7 님,

Doosan ROS Melodic 버전은 더 이상 공식적으로 지원하지 않습니다.
ROS1 사용자의 경우 Noetic 환경에서 사용하는 것을 부탁드립니다. 

사용 요구 환경은 다음과 같습니다.

- Ubuntu 20.04 LTS
- amd 64bit
- ROS1 Noetic / ROS2 Foxy

해당 환경이 맞지 않으면 DRCF Emulator 는 제대로 동작하지 않을 수 있습니다. 

감사합니다.

doosan-robotics, 2024-02-26:
@summerery7 님 

답변 감사합니다. 
보내주신 로그를 살펴본 결과 dsr01 으로 스폰하신 프로세스가 완전히 종료되지 않아서 발생하는 문제로 보입니다. 

[ERROR] [1708029214.948697009, 1614.516000000]: SpawnModel: Failure - model name dsr01 already exist.
 
해당 상황에서 ps ax 에서 gz 또는 이전에 실행하신  launcher 파일 명령어의 프로세스를 종료 후 다시한번 시도 부탁드립니다.

doosan-robotics, 2024-03-08:
@summerery7 님

안녕하세요 해당 이슈에 일정 기간 동안 답변이 없어 자동 종결 처리 하겠습니다. 

감사합니다.

closed 2024-03-08 by doosan-robotics