DoosanRobotics/doosan-robot #184: M0609 connection error

pbs0730, 2024-02-21:
M0609 모델로 우분투에서
roslaunch dsr_launcher single_robot_rviz.launch host:=192.168.0.22 port:=12345 mode:=real model:=m0609
를 이용해 원격제어를 시도했지만 짧으면 10초에서 길면 2분 이내에 로봇의 제어권이 회수됩니다.

그리고 추가로 우분투 터미널에서는 아래와 같은 경고문도 같이 나옵니다.
[WARN] [1708497749.233460]: Controller Spawner couldn't find the expected controller_manager ROS interface.

문제의 발생 원인이 버전의 문제인지, 인터넷의 문제인지 알고 싶습니다.
우분투 버전은 16.04.7 LTS, ROS 버전은 ROS1, kinetic 1.12.17
이렇게 사용하고 있습니다.

doosan-robotics, 2024-02-22:
안녕하세요 @pbs0730 님,

제어권이 회수되는 경우는 대체로 제어기와 ros pc 간의 연결이 불안정하거나 연결 시도 후 실패 했을때 발생됩니다. 추가적으로 터미널 경고문도 연결이 되지 않거나 중도에 끊겼음에 따라 ros interface에 응답이 없어서 발생하는 경고 문구로 보여집니다.

사용하신 명령어 상  Real 모드로 설정 되있는 것을 보았을때 host 설정은 로봇제어기 Ip 인 192.168.137.100 으로 연결 하셔야 하며 사용중인 ros pc 의 네트워크 설정도 이에 맞추어 필요합니다.

pbs0730, 2024-02-22:
빠른 답변 감사합니다. @doosan-robotics 님

답변 주신 대로 ip 설정을 192.168.137.100 으로 설정해 연결해보았을 때 처음 연결을 실패하는 경우가 현저히 줄어들었습니다.
감사합니다.

하지만 이후 발생하는 제어권이 회수되는 상황은 1분 정도 이후에 똑같이 발생합니다. 아니면 혹시 연결을 하고 작동을 시키지 않으면 자동으로 제어권이 회수되는 설정이 되어있는 건가요?

pbs0730, 2024-02-22:
주절주절 말이 많아지는 것 같아 여기에 정리하겠습니다. 죄송합니다.

중간에 host ip와 제어기 ip를 다르게 했던 경우, 원격 제어가 20분 이상 끊기지 않고 계속 유지됐지만, pc를 껐다 켜니 연결 자체가 안 됐습니다.
현재는 host ip와 제어기 ip를 동일하게 설정한 뒤 원격 제어를 시도하니 100% 처음 연결은 성공합니다. 하지만 젤 처음에 질문드린 것처럼 느려도 2분 이내에 제어권이 회수됩니다.

doosan-robotics, 2024-02-26:
@pbs0730 님 

답변 감사합니다.

기본적으로 host ip 의 input 값은 반드시 제어기 ip 와 동일해야만 연결이 가능합니다 . (제어기와 tcp 통신 연결을 위한 커멘드이기 때문입니다. )
20분 동안 끊기지 않고 유지 되었다고 하셨던 상황에서 로그를 보시면 initializing 를 계속 시도하는 상태의 20분일 가능성이 높습니다.

동일하게 설정한되에 연결이 성공 뒤에 제어권이 회수 된다는 점은 2가지 경우가 있습니다.
1. 제어기와 연결이 불안정 한 경우
2. 제어기에 tp 나 dart studio 등의 다른 곳에서 제어권이 회수를 시도할 경우 
입니다. 

pc와 제어기의 연결을 위해 네트워크 환경은 어떻게 사용하고 계신가요 ?
또한  /home/ampl/.ros/log/666944fe-d149-11ee-9d33-80d21dea4ae1/dsr01m0609-3*.log 에 생성된 로그도 함께 공유해주시면 분석에 참고하여 보다 자세한 답변을 드릴 수 있을 것 같습니다.

감사합니다.

pbs0730, 2024-02-27:
@doosan-robotics 님, 다시 한 번 답변에 감사드립니다.

1. 네트워크는 무선 인터넷 와이파이를 이용하고 있습니다.
2. 아래는 생성된 로그입니다.
[rospy.client][INFO] 2024-02-27 15:58:42,737: init_node, name[/dsr01/controller_spawner], pid[4671]
[xmlrpc][INFO] 2024-02-27 15:58:42,738: XML-RPC server binding to 127.0.0.1:0
[rospy.init][INFO] 2024-02-27 15:58:42,738: ROS Slave URI: [http://localhost:38127/]
[xmlrpc][INFO] 2024-02-27 15:58:42,738: Started XML-RPC server [http://localhost:38127/]
[rospy.impl.masterslave][INFO] 2024-02-27 15:58:42,739: _ready: http://localhost:38127/
[xmlrpc][INFO] 2024-02-27 15:58:42,739: xml rpc node: starting XML-RPC server
[rospy.registration][INFO] 2024-02-27 15:58:42,739: Registering with master node http://localhost:11311
[rospy.init][INFO] 2024-02-27 15:58:42,839: registered with master
[rospy.rosout][INFO] 2024-02-27 15:58:42,839: initializing /rosout core topic
[rospy.rosout][INFO] 2024-02-27 15:58:42,846: connected to core topic /rosout
[rospy.simtime][INFO] 2024-02-27 15:58:42,849: /use_sim_time is not set, will not subscribe to simulated time [/clock] topic
[rosout][INFO] 2024-02-27 15:58:42,854: Controller Spawner: Waiting for service controller_manager/load_controller
[rospy.internal][INFO] 2024-02-27 15:58:43,082: topic[/rosout] adding connection to [/rosout], count 0
[rosout][WARNING] 2024-02-27 15:59:13,014: Controller Spawner couldn't find the expected controller_manager ROS interface.
[rospy.core][INFO] 2024-02-27 15:59:13,014: signal_shutdown [atexit]
[rospy.internal][INFO] 2024-02-27 15:59:13,018: topic[/rosout] removing connection to /rosout
[rospy.impl.masterslave][INFO] 2024-02-27 15:59:13,018: atexit

doosan-robotics, 2024-02-28:
안녕하세요 @pbs0730 님,

해당 로그를 보아 connection 이 되지 않아 spawner에 필요한 state가 올라오지 않는 것으로 보입니다.
무선으로 연결했을 때 와이파이 신호가 약해지는 것 혹은 ip 충돌 등으로 연결이 되지 않거나 끊기는 현상이 발생합니다. 
- 일단 확인을 위해서 이더넷 케이블로 유선으로 로봇과 연결하여 한번 테스트를 해보시고 무선상태와 비교하는 것이 필요할 것으로 보입니다.
- 무선으로 잡으셨을 때는 라우터 기기를 사용 하셨을꺼라 짐작됩니다만 해당 장비의 ip 설정이나 gateway 설정등은 제어기에 맞게 설정하셨는지 확인 부탁드립니다

pbs0730, 2024-02-29:
@doosan-robotics 님, 제가 사용한 장비의 ip, gateway 설정에 뭔가 잘못된 게 있는지 확인해주실 수 있으실까요?

제어기
ip: 192.168.137.100
게이트웨이: 192.168.137.1

host ip : 192.168.137.100

원격 제어 pc 와이파이
ip: 192.168.137.50
게이트웨이: 192.168.137.1

doosan-robotics, 2024-03-05:
@pbs0730 님 
보여주신 네트워크 설정 상으로는 문제가 없습니다 .
유선 연결시에 init 연결 상태가 어떠한지 확인 부탁드립니다.

pbs0730, 2024-03-07:
@doosan-robotics 님. 정말 죄송하지만 init 연결 상태는 어디서 확인할 수 있을까요?

closed 2024-03-07 by pbs0730

duygiang1162, 2024-03-30:
How did you fixed this problem?