DoosanRobotics/doosan-robot #187: rosservice call /dsr01m1013/motion/move_joint not executing

zionh1, 2024-03-09:
안녕하십니까 현재 roslaunch dsr_launcher single_robot_gazebo.launch 작동 할때 에뮬레이터는 사전에 다 실행 해 놨고 
rosservice call /dsr01m1013/motion/move_joint "pos: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
vel: 0.0
acc: 0.0
time: 0.0
radius: 0.0
mode: 0
blendType: 0
syncType: 0"  작동을 할때 조인트 3번만 각도 조절이 되고 있습니다. 
lim@lim:~$ rosservice call /dsr01m1013/motion/move_joint "pos: [0.0, 0.0, 90.0, 0.0, 90.0, 0.0]
vel: 0.0
acc: 0.0
time: 3.0
radius: 0.0
mode: 0
blendType: 0
syncType: 0" 
success: True
[image]
해당 오류를 해결하기 위해 제가 할 수 있는 것이 있을까요? 
그리고 현재 m1013과 husky 모델의 결합된 모델을 불러 오는 것이 안되고 있습니다. 
현재는 husky 모델을 못 불러 오는 것인지 여쭤 보고싶습니다.

읽어 주셔서 감사합니다

doosan-robotics, 2024-03-11:
@zionh1 안녕하세요 

먼저 관심 갖고 사용해주셔서 감사합니다. 
해당 축이 움직이지 않는 건 pid 값을 사용자 편의에 따라서 gazebo yaml 값으로 조정할 수 있도록 옵션을 설정 되어있기 때문으로 파악됩니다. 
해당 옵션을 끄고 현재 로봇과 똑같은 움직임을 원하신다면 dsr_gazebo/launch/dsr_base.launch 에 아래 코드 부분을 주석 처리후에 다시 재시도를 하시면 실제 로봇과의 똑같은 모션으로 움직이는 것을 확인 할 수 있습니다. 

`  <group ns="$(arg ns)">
    <rosparam file="$(find dsr_gazebo)/config/gazebo_ros_control_params.yaml" command="load"/>
  </group>`

두번째로 질문 주신 husky 관련은 moblie 은 현재 지원하고 있지 않습니다.

zionh1, 2024-03-12:
답변 정말 감사합니다. 잘 작동되는 것을 확인했습니다.

closed 2024-03-12 by zionh1