DoosanRobotics/doosan-robot #188: Determine if a position is reachable

jwwood17, 2024-03-11:
Hello,
I'm trying to calculate positions to send the robot to, but sometimes the positions end up not being reachable, resulting in the following error
```
[ WARN] [1710177127.308328911]: [callback OnLogAlarm]
[ WARN] [1710177127.308354527]:  level : 2
[ WARN] [1710177127.308359402]:  group : 2
[ WARN] [1710177127.308363230]:  index : 1206
[ WARN] [1710177127.308367647]:  param : [ERR] Pose(-675.69mm[-26.6021in], -350.93mm[-13.8163in], 795.67mm[31.3255in], 89.98deg, -89.49deg, 13.24deg) is NOT REACHABLE
```
I would like to check whether the position is reachable before commanding the robot to go there, so I can filter out those invalid positions in the python script. Formally, I'd like to be able to use some function like
```
check_reachable(posx(50,200,150,135,-180,0),ref=2,sol=2)
```
returning true or false depending on whether the robot can reach that position. Is there any way to achieve this functionality using the ros package or the c++ api?

song-ms, 2024-03-12:
Dear @jwwood17,

If you want to ensure that the robot reaches a pose it can actually achieve before executing the actual motion, I believe the ikin() function would be appropriate for that purpose. the ikin() function is returns the joint position corresponding to sol_space, which is equivalent to the robot pose in the operating space, among 8 joint shapes.Through status of return value, you can check whether current robot pose is in wrist singularity or out of operating region. 
In ROS API, you can check status as the ikin_ex function.

Best Regards

jwwood17, 2024-03-12:
Thank you,
is there a way to check if a position would result in a collision with a safety zone? Also, I can't find any documentation for ikin_ex, what is the ref_pos_opt parameter for?

song-ms, 2024-03-13:
Dear @jwwood17,

Safety-related commands are currently not supported due to doosan API policies. Regarding zone settings, we recommend using the teaching pendant to configure them beforehand.

"ref_pos_opt" in Ikin_ex function determines closest joint position depend on option among multi turn solutions
0 : posj(0,0,0,0,0,0) is reference 
1 : current joint position is reference

Thank you

closed 2024-05-16 by jwwood17