DoosanRobotics/doosan-robot #193: No p gain specified for PID. Namespace: /dsr01/gazebo_ros_control/pid_gain/joint...

duygiang1162, 2024-03-27:
I launched command: roslaunch dsr_launcher single_robot_rviz_gazebo.launch mode:=virtual model:=m0609 color:=blue
But get error. Well i think this launch file didn't load file YAML above so i add new line in launch file as:  
 <rosparam file="$(find dsr_gazebo)/config/gazebo_ros_control_params.yaml" command="load"/>
This error is ongoing. Please give me advance to fix this problem

I using ubuntu 20.04 noetic distro
```
[ INFO] [1711532959.202417068, 0.369000000]: Loading gazebo_ros_control plugin
[ INFO] [1711532959.202510101, 0.369000000]: Starting gazebo_ros_control plugin in namespace: /dsr01
[ INFO] [1711532959.202957667, 0.369000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ERROR] [1711532959.313338525, 0.369000000]: No p gain specified for pid.  Namespace: /dsr01/gazebo_ros_control/pid_gains/joint1
[ERROR] [1711532959.313738387, 0.369000000]: No p gain specified for pid.  Namespace: /dsr01/gazebo_ros_control/pid_gains/joint2
[ERROR] [1711532959.314193148, 0.369000000]: No p gain specified for pid.  Namespace: /dsr01/gazebo_ros_control/pid_gains/joint3
[ERROR] [1711532959.314692259, 0.369000000]: No p gain specified for pid.  Namespace: /dsr01/gazebo_ros_control/pid_gains/joint4
[ERROR] [1711532959.315118453, 0.369000000]: No p gain specified for pid.  Namespace: /dsr01/gazebo_ros_control/pid_gains/joint5
[ERROR] [1711532959.315455455, 0.369000000]: No p gain specified for pid.  Namespace: /dsr01/gazebo_ros_control/pid_gains/joint6
[ INFO] [1711532959.318294231, 0.369000000]: Loaded gazebo_ros_control.
[INFO] [1711532959.418224, 0.380000]: Loading controller: dsr_joint_publisher
[INFO] [1711532959.428834, 0.391000]: Controller Spawner: Loaded controllers: dsr_joint_position_controller, dsr_joint_publish
```er

duygiang1162, 2024-04-04:
Thank you for your feedback. I tried with the pid parameters you provided. But the robot in the simulation keeps shaking. Is this because these parameters have not been optimized or is there another reason that I don't know yet? And does your simulation have the same problem as mine?

duygiang1162, 2024-04-04:
I discovered that, without using pid parameters, the robot operates more stably in the simulation environment. Is this a good way to continue developing robots?

vyasakash231, 2024-04-05:
> Thank you for your feedback. I tried with the pid parameters you provided. But the robot in the simulation keeps shaking. Is this because these parameters have not been optimized or is there another reason that I don't know yet? And does your simulation have the same problem as mine?

Yes, I also experienced the same error, but after this step, It was gone.

vyasakash231, 2024-04-05:
> I discovered that, without using pid parameters, the robot operates more stably in the simulation environment. Is this a good way to continue developing robots?

not sure; I'm also new to this.

vyasakash231, 2024-04-05:
@duygiang1162, please let me know whether you were able to fix the Rviz issue.