DoosanRobotics/doosan-robot #202: Topic associated to LogAlarm.msg not existing

buswayne, 2024-05-02:
There is no topic associated to the dsr_msgs/LogAlarm.msg type, though the corresponding service (GetLastAlarm.srv) it does.

song-ms, 2024-05-10:
Dear @buswayne,

Thank you for the valuable input.
However, creating log alert messages could potentially lead to an increase in topic volume, especially in scenarios like real-time commands, which might cause communication issues. Therefore, we offer monitoring for errors or specific events within the logs or read last alarm as ros service type.

Thank you

buswayne, 2024-06-06:
Thank you,

I understand, but adding it to the "/state" topic would be useful with minimum impact (even just the logalarm code is useful). Some situations might require constant monitoring of the log alarm to provide specific actions (like in our case)

R

closed 2024-06-06 by buswayne