DoosanRobotics/doosan-robot #203: Lower Collision Threshold

jwwood17, 2024-05-02:
Hello,
Is there any way to lower the robot's collision threshold temporarily in a script? I have a motion I want to accomplish with compliance control, but it keeps registering a collision. I don't want to adjust these values in the teach pendant because I want to preserve safety measures outside of this one movement.
Thank you

song-ms, 2024-05-10:
Hello, @jwwood17 

In accordance with our safety policy, the collision threshold setting parameter is classified under safety parameters and is not provided through the API. Therefore, for dynamic adjustment of the collision threshold, we recommend pre-setting values within a safety zone and utilizing them accordingly.

Thank you.

jwwood17, 2024-05-10:
Hello, @song-ms

What do you mean by "dynamic adjustment of the collision threshold" and "utilizing them accordingly?" Is there a way to dynamically change active safety zones? How would you suggest I proceed for lowering the collision sensitivity for a single can't just use a safety zone because I want to detect any collisions that occur in that area, just not during this movement.
Thank you.

song-ms, 2024-05-13:
Hello @jwwood17 ,

I mean that value of collision sensitivity cannot be technically changed on a single movement basis by safety policy. As pre-setting the collision sensitivity in safety zone on Teaching Pendent before running the program, levels of collision sensitivity can be changed by pre-setting basis of area.

Thank you