DoosanRobotics/doosan-robot #208: TCP and TOOL setting issues in ROS 1

Arpit-Bhat, 2024-06-11:
1. I set and enable the Tool setting in Teach Pendent, then I try to get current TCP feedback in ROS environment, and it shows the current set TCP, but when I disconnect and reconnect the ROS master and control launch file, it automatically disable the TOOL setting in Teach Pendent. Moreover, when I kill the control launch file , the Tool setting in teach pendant also disabled
2. Now, I try to use the following command to create a TCP or TOOL, and I also try to set the TCP and TOOL, but it shows an error index of 7170, hence the following commands are not working.
     1. ConfigCreateTcp.srv
     2. SetCurrentTcp.srv
     4. ConfigCreateTool.srv
     5. SetCurrentTool.srv

Master node responce
[ INFO] [1718016769.336561224]: < get_current_posx_cb >
[ WARN] [1718017058.862818040]: [callback OnLogAlarm]
[ WARN] [1718017058.862898980]:  level : 2
[ WARN] [1718017058.862958767]:  group : 5
[ WARN] [1718017058.863041427]:  index : 7170
[ WARN] [1718017058.863106249]:  param : 1
[ WARN] [1718017058.863183053]:  param : 
[ WARN] [1718017058.863538991]:  param : 
[ WARN] [1718017058.866212921]: [callback OnLogAlarm]
[ WARN] [1718017058.866293947]:  level : 2
[ WARN] [1718017058.866373624]:  group : 5
[ WARN] [1718017058.866439583]:  index : 7170
[ WARN] [1718017058.866502957]:  param : 0
[ WARN] [1718017058.866589667]:  param : 
[ WARN] [1718017058.866682349]:  param :

Arpit-Bhat, 2024-06-20:
The given sample file for Tool is not working for model a0509

Alessandro-Colombo, 2024-10-17:
I have the same exact error, trying to change the TOOL on a model m1013

Alessandro-Colombo, 2024-10-17:
I eventually managed to do that, by setting robot mode to 0 (handguiding).
I'm not sure if this is the intended way to do it.
It didn't work in robot mode 1.