DoosanRobotics/doosan-robot #210: Achieving smooth motion on hardware

JaredRosenbaum, 2024-06-24:
Hello,

1. While controlling our H2017 with the doosan-robot package, we've been noticing that often the robot will move in an uneven manner, with alternating sections of fast and slow motion as seen in [this video](https://imgur.com/a/nGvv23w), of the robot moving from one setpoint to another using the dsr_moveit launch file. Is this a known issue, or is there a known fix for a mistake we might be making?

2. Additionally, we're hoping to control our arm using [moveit servo](https://index.ros.org/p/moveit_servo/). Our current setup sees the servo node publish commands as trajectory_msgs/JointTrajectory, which we then send to the FollowJointTrajectory Action server as an [action goal](https://docs.ros.org/en/electric/api/control_msgs/html/msg/FollowJointTrajectoryActionGoal.html). Although the commands are published at a rate of 100hz, the motion by the arm is piecewise as seen in [this video](https://imgur.com/a/EuCqdfp). Our initial guess is that this occurs because each movement command blocking and must complete before the next command begins; is this intuition correct?

doosan-robotics, 2024-07-10:
Hello @JaredRosenbaum 

Regarding the first question, we are aware of the issue internally and it is scheduled to be addressed with an improved motion servoj in the upcoming 2.12 version update.

As for the second question about the servo, it pertains to the moveit_servo functionality. Since our provided package does not utilize the moveit_servo feature, it may be challenging to resolve everything with the improved servoj motion.

Thank you

doosan-robotics, 2024-08-06:
Hello @JaredRosenbaum ,

If you would like to test a version faster than 2.12, we can provide you with the developer version 2.11.2.1. 
Please review and let us know.

Thank you

JaredRosenbaum, 2024-08-19:
Hello,
We would kindly appreciate the opportunity to work with the development version. How can we go about getting this?

Thank you