DoosanRobotics/doosan-robot #217: Docker setup for real robot control

jwwood17, 2024-07-30:
[compose.yml](https://github.com/user-attachments/files/16433254/compose.txt)
[Dockerfile](https://github.com/user-attachments/files/16433255/Dockerfile.txt)
I am trying to control an m1013 robot arm on my windows machine using a docker container running in Ubuntu 20.04 with ROS noetic. I've attached my Dockerfile and compose.yml.
When I run the following code, I get several errors as the connection fails on an assertion in the dsr_hw_interface.cpp file.
```
roslaunch dsr_control dsr_control.launch mode:=real model:=m1013 host:=206.207.65.90
```

```
started roslaunch server http://35b1eea02359:40149/

SUMMARY
========

PARAMETERS
 * /dsr01m1013/command: True
 * /dsr01m1013/dsr_joint1_position/joint: joint1
 * /dsr01m1013/dsr_joint1_position/pid/d: 200.0
 * /dsr01m1013/dsr_joint1_position/pid/i: 100.0
 * /dsr01m1013/dsr_joint1_position/pid/i_clamp: 500
 * /dsr01m1013/dsr_joint1_position/pid/p: 500.0
 * /dsr01m1013/dsr_joint1_position/type: position_controll...
 * /dsr01m1013/dsr_joint2_position/joint: joint2
 * /dsr01m1013/dsr_joint2_position/pid/d: 250.0
 * /dsr01m1013/dsr_joint2_position/pid/i: 100.0
 * /dsr01m1013/dsr_joint2_position/pid/i_clamp: 500
 * /dsr01m1013/dsr_joint2_position/pid/p: 500.0
 * /dsr01m1013/dsr_joint2_position/type: position_controll...
 * /dsr01m1013/dsr_joint3_position/joint: joint3
 * /dsr01m1013/dsr_joint3_position/pid/d: 40.0
 * /dsr01m1013/dsr_joint3_position/pid/i: 20.0
 * /dsr01m1013/dsr_joint3_position/pid/i_clamp: 500
 * /dsr01m1013/dsr_joint3_position/pid/p: 300.0
 * /dsr01m1013/dsr_joint3_position/type: position_controll...
 * /dsr01m1013/dsr_joint4_position/joint: joint4
 * /dsr01m1013/dsr_joint4_position/pid/d: 40.0
 * /dsr01m1013/dsr_joint4_position/pid/i: 0.0
 * /dsr01m1013/dsr_joint4_position/pid/i_clamp: 500
 * /dsr01m1013/dsr_joint4_position/pid/p: 100.0
 * /dsr01m1013/dsr_joint4_position/type: position_controll...
 * /dsr01m1013/dsr_joint5_position/joint: joint5
 * /dsr01m1013/dsr_joint5_position/pid/d: 25.0
 * /dsr01m1013/dsr_joint5_position/pid/i: 20.0
 * /dsr01m1013/dsr_joint5_position/pid/i_clamp: 500
 * /dsr01m1013/dsr_joint5_position/pid/p: 200.0
 * /dsr01m1013/dsr_joint5_position/type: position_controll...
 * /dsr01m1013/dsr_joint6_position/joint: joint6
 * /dsr01m1013/dsr_joint6_position/pid/d: 25.0
 * /dsr01m1013/dsr_joint6_position/pid/i: 20.0
 * /dsr01m1013/dsr_joint6_position/pid/i_clamp: 500
 * /dsr01m1013/dsr_joint6_position/pid/p: 300.0
 * /dsr01m1013/dsr_joint6_position/type: position_controll...
 * /dsr01m1013/dsr_joint_position_controller/joints: ['joint1', 'joint...
 * /dsr01m1013/dsr_joint_position_controller/publish_rate: 20
 * /dsr01m1013/dsr_joint_position_controller/type: position_controll...
 * /dsr01m1013/dsr_joint_publisher/publish_rate: 150
 * /dsr01m1013/dsr_joint_publisher/type: joint_state_contr...
 * /dsr01m1013/dsr_joint_trajectory_controller/joints: ['joint1', 'joint...
 * /dsr01m1013/dsr_joint_trajectory_controller/publish_rate: 20
 * /dsr01m1013/dsr_joint_trajectory_controller/type: position_controll...
 * /dsr01m1013/dsr_velocity_controller/angular/z/has_acceleration_limits: True
 * /dsr01m1013/dsr_velocity_controller/angular/z/has_velocity_limits: True
 * /dsr01m1013/dsr_velocity_controller/angular/z/max_acceleration: 6.0
 * /dsr01m1013/dsr_velocity_controller/angular/z/max_velocity: 2.0
 * /dsr01m1013/dsr_velocity_controller/base_frame_id: base_link
 * /dsr01m1013/dsr_velocity_controller/cmd_vel_timeout: 0.25
 * /dsr01m1013/dsr_velocity_controller/enable_odom_tf: False
 * /dsr01m1013/dsr_velocity_controller/estimate_velocity_from_position: False
 * /dsr01m1013/dsr_velocity_controller/left_wheel: ['front_left_whee...
 * /dsr01m1013/dsr_velocity_controller/linear/x/has_acceleration_limits: True
 * /dsr01m1013/dsr_velocity_controller/linear/x/has_velocity_limits: True
 * /dsr01m1013/dsr_velocity_controller/linear/x/max_acceleration: 3.0
 * /dsr01m1013/dsr_velocity_controller/linear/x/max_velocity: 1.0
 * /dsr01m1013/dsr_velocity_controller/pose_covariance_diagonal: [0.001, 0.001, 0....
 * /dsr01m1013/dsr_velocity_controller/publish_rate: 50
 * /dsr01m1013/dsr_velocity_controller/right_wheel: ['front_right_whe...
 * /dsr01m1013/dsr_velocity_controller/twist_covariance_diagonal: [0.001, 0.001, 0....
 * /dsr01m1013/dsr_velocity_controller/type: diff_drive_contro...
 * /dsr01m1013/dsr_velocity_controller/wheel_radius_multiplier: 1.0
 * /dsr01m1013/dsr_velocity_controller/wheel_separation_multiplier: 1.875
 * /dsr01m1013/gripper: none
 * /dsr01m1013/host: 206.207.65.90
 * /dsr01m1013/mobile: none
 * /dsr01m1013/mode: real
 * /dsr01m1013/model: m1013
 * /dsr01m1013/moveit: False
 * /dsr01m1013/name: dsr01
 * /dsr01m1013/port: 12345
 * /dsr01m1013/rate: 100
 * /dsr01m1013/standby: 5000
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    dsr01m1013 (dsr_control/dsr_control_node)
  /dsr01m1013/
    controller_spawner (controller_manager/spawner)

ROS_MASTER_URI=http://localhost:11311

process[dsr01m1013/controller_spawner-1]: started with pid [160]
process[dsr01m1013-2]: started with pid [161]
[INFO] [1722370041.055621]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1722370041.842006909]: rate is 100

[ INFO] [1722370041.848552455]: name_space is dsr01, m1013

[ INFO] [1722370041.848595859]: [dsr_hw_interface] constructed
[ INFO] [1722370041.884660080]: [dsr_hw_interface] init() ==> setup callback fucntion
[ INFO] [1722370041.884697848]: INIT@@@@@@@@@@@@@@@@@@@@@@@@@
[ INFO] [1722370041.884721184]: [dsr_hw_interface] init() ==> arm is standby
[ INFO] [1722370041.885442601]: host 206.207.65.90, port=12345 bCommand: 1, mode: real

dsr_control_node: /root/catkin_ws/src/doosan-robot/dsr_control/src/dsr_hw_interface.cpp:1137: bool dsr_control::DRHWInterface::init(): Assertion `Drfl.get_system_version(&tSysVerion)' failed.
[dsr01m1013-2] process has died [pid 161, exit code -6, cmd /root/catkin_ws/devel/lib/dsr_control/dsr_control_node /dsr_joint_position_controller/command:=/dsr01/dsr_joint_position_controller/command __name:=dsr01m1013 __log:=/root/.ros/log/3fc4ace2-4eaf-11ef-a3a6-0242ac110002/dsr01m1013-2.log].
log file: /root/.ros/log/3fc4ace2-4eaf-11ef-a3a6-0242ac110002/dsr01m1013-2*.log
[WARN] [1722370344.476341]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[dsr01m1013/controller_spawner-1] process has finished cleanly
log file: /root/.ros/log/3fc4ace2-4eaf-11ef-a3a6-0242ac110002/dsr01m1013-controller_spawner-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
I have properly set the ip address of the robot to 206.207.65.90 on the teach pendant and my windows machine is 206.207.65.91 on the same network. I can connect to the robot successfully on our linux computer, but not through the docker container. I am able to ping the robot through both the container and the host computer.
This is a screenshot of the captured traffic during the roslaunch call in wireshark.
[image]

It is worth noting that I've had a similar issue in trying to connect to the robot through a virtual machine that I also have not been able to solve. https://github.com/doosan-robotics/doosan-robot/issues/164

Any assistance in getting this to work would be greatly appreciated. 20.04 is quickly becoming an outdated operating system and I would like to be able to control the robot from a more up-to-date environment.

jwwood17, 2024-08-23:
> @jwwood17 When you start the container, have to set the the network as `host`? `--net host`

Unfortunately I get the same issue. I tried both with host and bridge through docker run and through docker compose. I also tried running docker in wsl instead of windows, but I always get the same error. I can always ping the robot successfully, but it won't connect.

jwwood17, 2024-08-23:
Oddly enough, it did work when I ran the same docker container on an ubuntu 20.04 computer. That doesn't solve the problem, but maybe it helps diagnose it. It's ironic and strange that the docker container is performing differently on two different hosts.

gautham-nf, 2024-08-24:
Then this would be a connection issue from the windows side., you can check the windows firewall.

jwwood17, 2024-08-26:
@gautham-nf That's a good thought. I have tried temporarily disabling the firewall and it didn't help, but maybe it could have something to do with the docker build. To my understanding docker on windows runs containers inside a linux vm. Maybe that's the issue. I'll try to make a windows platform image and see if I can get it to run natively and if that works or not.