DoosanRobotics/doosan-robot #235: How to get force-torque sensor from Doosan A0509s

vyasakash231, 2025-05-18:
Please help @doosan-robotics. I tried to plot force-torque sensor readings, but I'm getting confused between `raw_force_torque`, `external_tcp_force` and `external_joint_torque`. Which one of these is an external wrench?
Furthermore, when the robot is operated in real time, why are these forces or torques varying, even though I'm not applying any external wrench (Fx, Fy, Fz, Mx, My, Mz)?

leeminju531, 2025-05-19:
Hi, `external` prefix indicates disturbances.
You can refer to a similar issue: https://github.com/DoosanRobotics/doosan-robot2/issues/203.

leeminju531, 2025-05-20:
> but even if I'm not applying any force on the end-effector, why am I getting 20-40 N of force, and it's varying as my robot moves?

1. Could you clarify what the values `raw_force_torque`, `external_tcp_force`, and `external_joint_torque` represent during robot motion?

2. In your setup, did you attach any additional end-effector above `link6`?

vyasakash231, 2025-05-20:
`raw_force_torque`, `external_tcp_force`, and `external_joint_torque` are from from the msg `RobotStateRT.msg`, https://github.com/DoosanRobotics/doosan-robot/blob/ab24968ebbc9e5f813d673adcf232c1586720dcb/dsr_msgs/msg/RobotStateRT.msg#L25-L30
I'm also trying to figure out what these values represent. I was looking for a way to measure external wrench applied at the end-effector.

Just to give you an example, I have performed trajectory tracking using operation space control. As you can see, the `raw_force_torque` & `external_tcp_force` are identical, but their values are varying with time, even though I've not applied any external wrench.

[image]

Also, there is no gripper or other attachments at the end.

doosan-robotics, 2025-05-22:
@vyasakash231, Could you attach the graph without moving it ?

vyasakash231, 2025-05-23:
@doosan-robotics, this is the result with gravity compensation at `[0.0, 0.0, 0.0, 0.0, 0.0,0.0]` joint position (vertically standing position).

[image]

vyasakash231, 2025-05-23:
@doosan-robotics, how can we calibrate the force-torque sensor?

elton-choi, 2025-06-14:
The difference between raw_force_torque and external_force_torque is just filtering. But they have same offset and scale. raw_force_torque gives you non-filtered data from sensor(but calibrated), but we filtering external_force_torque to show you stable value.

If you don't set tool weight, but they have offset in zero position.
I think you should calibrate fts sensor again in dart studio.
In service menu, you can find fts calibration function.

vyasakash231, 2025-06-17:
@elton-choi, So I've installed Dart Studio on my laptop, but I cannot find the service menu. Can you please help? I've never used Dart Studio.

[image]

closed 2025-06-27 by vyasakash231