DoosanRobotics/doosan-robot #236: Getting jacobian

PARKBONG, 2025-05-23:
Hi. I'm trying to get a Jacobian matrix to test my custom controller. 

what is the best way to access the internal jacobian matrix?

I found the [Jacobian matirx](https://github.com/DoosanRobotics/doosan-robot/blob/ab24968ebbc9e5f813d673adcf232c1586720dcb/dsr_control/src/dsr_hw_interface.cpp#L3337) in the dsr_hw_interface.cpp, which is advertised as hidden topic as [here](https://github.com/DoosanRobotics/doosan-robot/blob/ab24968ebbc9e5f813d673adcf232c1586720dcb/dsr_control/src/dsr_hw_interface.cpp#L1096). 

I tried to look for some docs/apis, but I cannot find any infomation.
What is the best way to access the internal Jacobian matrix?

PARKBONG, 2025-05-23:
I appreciate if you give me a step-by-step approaches, as below. 

# start robot
alias robot='roslaunch dsr_launcher dsr_moveit.launch model:=m1013'

# connect realtime controller
rosservice call /dsr01m1013/realtime/connect_rt_control "{ip_address: '192.168.1.100', port: 12347}"

# start rtc
rosservice call /dsr01m1013/realtime/start_rt_control "{}"

# read Jacobian
rosservice call /dsr01m1013/realtime/read_data_rt "{}" # which is unavailable since it is published on private node

PARKBONG, 2025-05-23:
main.cpp
```
#include <iostream>
#include <cstring>
#include <unistd.h>
#include "DRFL.h"
#include "DRFLEx.h"

using namespace std;
using namespace DRAFramework;

int main() {
    CDRFLEx drfl;

    cout << "[INFO] Connecting to RT control..." << endl;
    if (!drfl.connect_rt_control("192.168.137.100", 12345)) {
        cerr << "[ERROR] Failed to connect to RT control." << endl;
        return -1;
    }
    cout << "[INFO] Connected to RT control." << endl;

    cout << "[INFO] Setting RT output..." << endl;
    drfl.set_rt_control_output("v1.0", 0.001f, 4);

    cout << "[INFO] Starting RT control..." << endl;
    if (!drfl.start_rt_control()) {
        cerr << "[ERROR] Failed to start RT control." << endl;
        return -1;
    }
    cout << "[INFO] RT control started." << endl;

    for (int i = 0; i < 10; ++i) {
        cout << "[INFO] Waiting for RT data..." << endl;
        auto rt_data = drfl.read_data_rt();
        if (!rt_data) {
            cerr << "[ERROR] Failed to read RT data." << endl;
            continue;
        }

        cout << "[INFO] Jacobian Matrix (row-wise):" << endl;
        for (int r = 0; r < 6; ++r) {
            for (int c = 0; c < 6; ++c) {
                cout << rt_data->jacobian_matrix[r][c] << " ";
            }
            cout << endl;
        }

        usleep(100000);
    }

    drfl.stop_rt_control();
    drfl.disconnect_rt_control();
    cout << "[INFO] Finished." << endl;

    return 0;
}
```
Makefile
```
# ~/drfl_native_example/Makefile

CXX = g++
CXXFLAGS = -std=c++11 -Wall

INCLUDES = -I/home/smslab/catkin_ws/src/doosan-robot/common/include/
LIBS = /home/smslab/catkin_ws/src/doosan-robot/common/lib/noetic/x86_64/libDRFL.a

SRC = src/main.cpp
OUT = drfl_rt_demo

all:
	$(CXX) $(CXXFLAGS) $(SRC) -o $(OUT) $(INCLUDES) $(LIBS) -lPocoFoundation -lPocoUtil -lPocoNet -lpthread -ldl
```

Structure
~/drfl_native_example$ tree .
.
├── drfl_rt_demo
├── Makefile
└── src
    └── main.cpp
1 directory, 3 files

lanch @ terminal
```
$ roslaunch dsr_launcher dsr_moveit.launch model:=m1013
$ make 
$ ./drfl_rt_demo

[INFO] Connecting to RT control...
usClientPort : 59003
Connection refused
Close UDP Connection
[INFO] Connected to RT control.
[INFO] Setting RT output...
[INFO] Starting RT control...
[INFO] RT control started.
[INFO] Waiting for RT data...
[INFO] Jacobian Matrix (row-wise):
0 0 0 0 0 0 
0 0 0 0 0 0 
0 0 0 0 0 0 
0 0 0 0 0 0 
0 0 0 0 0 0 
0 0 0 0 0 0 
[INFO] Waiting for RT data...
[INFO] Jacobian Matrix (row-wise):
0 0 0 0 0 0 
0 0 0 0 0 0 
0 0 0 0 0 0 
0 0 0 0 0 0 
0 0 0 0 0 0 
0 0 0 0 0 0 
[INFO] Waiting for RT data...
[INFO] Jacobian Matrix (row-wise):
```

could you explain what I missed? thanks.

doosan-robotics, 2025-07-07:
Hello @PARKBONG ,

When using the package, the robot mode must be set to real in order to use real-time related APIs.
If the mode is not explicitly set to real, it will run in virtual mode by default

roslaunch dsr_launcher dsr_moveit.launch model:=m1013 host:=192.168.137.100 port:=12345
+ mode:=real

song-ms, 2025-07-15:
Dear @PARKBONG ,

When we conducted tests on our side, the system operated normally without any connection failures. We would like to confirm the environment in which you experienced the issue.

Please note that the client limit for starting the control connection is restricted to one user by default. Kindly check if the "start RT control" was declared redundantly.